adafruit/Adafruit_CircuitPython_Wiznet5k #15: Long int not supported on M0 CircuitPython

xorbit, 2020-05-10:
On an Adafruit Feather M0 Adalogger:

```python
>>> from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 49, in <module>
OverflowError: long int not supported in this build
```

Is there really no way to avoid using long int to make this work on M0 boards?

ladyada, 2020-05-10:
it really wont fit and anyways wiznet will not work either - there's just not enough memory to do internet things. can you try the feather m4?

xorbit, 2020-05-10:
Alright, thanks for confirming that.

Would it work on a Feather NRF52840?  I think that's an M4 and this board would be generally more useful to me with the BLE, but I can't find any in stock anywhere at the moment.

Can you point me to a place with stock?  If there isn't any, I'll order a Feather M4 and try with that.

ladyada, 2020-05-10:
yeah the nrf52840 will work. we aren't in stock now, you can sign up to be notified or try the feather m4 https://www.digikey.com/product-detail/en/adafruit-industries-llc/3857/1528-2648-ND/9553567

closed 2020-05-10 by ladyada