adafruit/Adafruit_CircuitPython_Wiznet5k #19: DNS resolv subdomain, Adafruit CircuitPython 5.3.0 on 2020-04-29

bytelinker, 2020-05-23:
There are errors resolving sub domains.

import board
import busio
import digitalio
import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket
cs = digitalio.DigitalInOut(board.D10)
spi_bus = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
eth = WIZNET5K(spi_bus, cs, is_dhcp=True)
requests.set_socket(socket, eth)
print("Chip Version:", eth.chip)
print("MAC Address:", [hex(i) for i in eth.mac_address])
print("My IP address is:", eth.pretty_ip(eth.ip_address))
print("My IP_ADDRESS is:", eth.pretty_ip(eth.ifconfig[0]))
print("My SUBNET_MASK is:", eth.pretty_ip(eth.ifconfig[1]))
print("My GATEWAY_ADDRESS is:", eth.pretty_ip(eth.ifconfig[2]))
print("My DNS_SERVER is:", eth.pretty_ip(eth.ifconfig[3]))

print('-'*40)

for i in ['adafruit.com','wikipedia.de','www.wikipedia.de','www.wikipedia.org','wikipedia.org']:
    try:
        ip=eth.pretty_ip(eth.get_host_by_name(i))
        print(f'IP lookup {i}: {ip}')
    except Exception as ex:
        print(i,ex)

Output:
#--------------------#
#
#Chip Version: w5500
#MAC Address: ['0xde', '0xad', '0xbe', '0xef', '0xfe', '0xed']
#My IP address is: 192.168.178.102
#My IP_ADDRESS is: 192.168.178.102
#My SUBNET_MASK is: 255.255.255.0
#My GATEWAY_ADDRESS is: 192.168.178.1
#My DNS_SERVER is: 192.168.178.1
#----------------------------------------
#IP lookup adafruit.com: 104.20.38.240
#IP lookup wikipedia.de: 134.119.24.29
#www.wikipedia.de Failed to resolve hostname!
#www.wikipedia.org Failed to resolve hostname!
#IP lookup wikipedia.org: 91.198.174.192

#Adafruit CircuitPython 5.3.0 on 2020-04-29

Thank you very much for you time.
bytelinker

brentru, 2020-06-08:
Does this still occur on the latest version of this library?

bytelinker, 2020-06-09:
Adafruit CircuitPython 5.4.0-beta.1 on 2020-06-08; Adafruit Grand Central M4 Express with samd51p20

adafruit_wiznet5k downloaded today from github

eth = WIZNET5K(spi_bus, cs, is_dhcp=True, hostname='otto', dhcp_timeout=35)

Hostname is visible in W2008R2 DHCP Server
code.py output:
* Received DHCP Message is not OFFER
* Received DHCP Message is not OFFER
r
Chip Version: w5500
MAC Address: ['0xxx', '0xxx', '0xxx', '0xxx', '0xxx', '0xxx']
My IP address is: xx.xx.xx.xx
My SUBNET_MASK is: 255.255.255.0
My GATEWAY_ADDRESS is: xx.xx.xx.xx
My DNS_SERVER is: xx.xx.xx.xx
----------------------------------------
IP lookup adafruit.com: 104.20.38.240
IP lookup wikipedia.de: 134.119.24.29
www.wikipedia.de Failed to resolve hostname!
www.wikipedia.org Failed to resolve hostname!
IP lookup wikipedia.org: 91.198.174.192

Still with problems.

bytelinker, 2020-06-09:
on a second network (other location, other switch and other router) I get the same errors.

bytelinker, 2020-08-13:
Are there any additional information needed, that I could provide?

brentru, 2020-08-14:
@bytelinker  I haven't seen this issue elsewhere on the 'net. Are you able to take a network capture (preferably `.pcap`) using Wireshark of the W5k DNS client attempting to access the subdomain? It would speed my debugging up.

brentru, 2020-09-25:
@bytelinker  Are you able to provide a network capture (.pcap file)?

kevinaj, 2020-09-28:
I believe this issue is the same as #24. `www.wikipedia.de` is a CNAME pointing to `wikipedia.de`

PR #73 (2023-01-20): Fix cname issue
closes #19 #24

1. Wrote some unit tests to make sure that my changes don't change the external behaviour of the DNS library, nor it's interaction with the socket to send and receive data.
2. Refactored the code to separate reading the DNS record from parsing the DNS record.
3. Refactored building the query and parsing the response into functions so that they can be unit tested without mocking out the WIZNET5K interface and the socket.
4. Refactored the DNS query builder to hardcode most of it, as only the question name and query ID vary.
5. Refactored the DNS response parser t [... cut]
closed 2023-01-20 by FoamyGuy