adafruit/Adafruit_CircuitPython_Wiznet5k #22: "Received DHCP Message is not OFFER" on OpenWRT ?

water5, 2020-09-17:
I connect wiznet5k to Mikrotik router, success to request IP, using:
`import wiznet5k_simpletest`

---------------------------------

Then connect wiznet5k to OpenWRT router, change `eth = WIZNET5K(spi_bus, cs, debug = True)` in wiznet5k_simpletest.py to enable debug message, receive message below:

`import wiznet5k_simpletest`
> Wiznet5k WebClient Test
>* Initializing DHCP
>*** Get socket
>Allocated socket #0
>* w5k socket connect, protocol=2, port=67, ip=255.255.255.255
>*** Opening socket 0
>* Opening W5k Socket, protocol=2
>* DHCP: Discover
>* DHCP: Parsing OFFER
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>* socket_available called with protocol 2
>Bytes avail. on sock:  308
>         * Processing 8 bytes of data
>* Received DHCP Message is not OFFER
>*** Closing socket #0
>Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "wiznet5k_simpletest.py", line 17, in <module>
>  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 196, in __init__
>AssertionError: Failed to configure DHCP Server!

Which scene will raise `Received DHCP Message is not OFFER` ? I ensure connect other device(e.g. PC) to this OpenWRT router can success request IP.
How to find what's going wrong?

water5, 2020-09-19:
I test `wiznet5k_simpletest_manual_network.py` with modified network configure below:
`IP_ADDRESS = (192, 168, 1, 3)`
`SUBNET_MASK = (255, 255, 255, 0)`
`GATEWAY_ADDRESS = (192, 168, 1, 1)`
`DNS_SERVER = (8, 8, 8, 8)`

then connect to same OpenWRT router, all result is fine, (router IP "`192.168.1.1`"):
`import wiznet5k_simpletest_manual_network`

> Wiznet5k WebClient Test (no DHCP)
> Chip Version: w5500
> MAC Address: ['0xde', '0xad', '0xbe', '0xef', '0xfe', '0xed']
> My IP address is: 192.168.1.3
> IP lookup adafruit.com: 104.20.38.240
> Fetching text from http://wifitest.adafruit.com/testwifi/index.html
>
> ----------------------------------------
> This is a test of Adafruit WiFi!
> If you can read this, its working :)

This's [Wireshark Capture file](https://github.com/water5/MicroPython_temporary/raw/master/wiznet5k_dhcp_cap.pcap) when test `wiznet5k_simpletest.py` with DHCP, it looks like already assigned IP `192.168.1.167` on OpenWRT router, I have no idea why it raise "`Received DHCP Message is not OFFER`".

kevinaj, 2020-09-20:
I have similar issues with OpenWRT when obtaining a lease for the first time. If there is an existing lease that hasn't expired, it seems to have no problem.

Increasing the `dhcp_timeout` worked for me: 
```eth = WIZNET5K(spi, cs, dhcp_timeout=10, debug=True)```

I also have a while loop that will keep trying until the network is connected and it obtains an address. This lets me connect the network cable after boot.

water5, 2020-09-21:
Which environment (board, version of CircuitPython, wiznet5k library) you are using?
If I specify `dhcp_timeout=10`, raise these error:

> Wiznet5k WebClient Test
> * Initializing DHCP
> *** Get socket
> Allocated socket #0
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "wiznet5k_simpletest.py", line 17, in <module>
>   File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 195, in __init__
>   File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 213, in set_dhcp
>   File "adafruit_wiznet5k/adafruit_wiznet5k_dhcp.py", line 364, in request_dhcp_lease
> ValueError: Random number generation error

water5, 2020-09-21:
OK, test pass after increase `dhcp_timeout` value, thanks.

But other issue raise, sometimes (not regular) receive this error:

> Wiznet5k WebClient Test
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "wiznet5k_simpletest.py", line 17, in <module>
>   File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 195, in __init__
>   File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 213, in set_dhcp
>   File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_dhcp.py", line 364, in request_dhcp_lease
> ValueError: Random number generation error

I try to generate random number independently, look like no problem:
`from random import randrange`
`for i in range(1000):`
`print(randrange(1, 2000))`

The `ValueError: Random number generation error` is it come from `random` module ? or `wiznet5k_dhcp.py` ?

kevinaj, 2020-09-23:
No problem. 

I'm using:
**Board:** Feather M4 Express
**CircuitPython:** 5.3.1
**Wiznet5k:** master, commit 18e9648f0bb4976ef5ebb6d92a10c173504eb3a8

I have not run into the random number generation error.

closed 2020-09-23 by water5

brentru, 2020-09-23:
> I use random.randint() replace random.randrange() in adafruit_wiznet5k_dhcp.py line 364, seems better, need more time to confirm.

@water5  What hardware are you using?