adafruit/Adafruit_CircuitPython_Wiznet5k #24: DNS lookup fails on CNAME records

kevinaj, 2020-09-25:
When a DNS query returns a CNAME record, the A record is never found. The example below uses [api.ipify.org](http://api.ipify.org)

**CircuitPython:** 5.3.1
**Adafruit_CircuitPython_Wiznet5k:** 1.5.0
**Hardware:** POE-Featherwing on Feather M4 Express

Source snippet:
```
eth = WIZNET5K(spi, cs, dhcp_timeout=10, debug=True)
requests.set_socket(socket, eth)
requests.get('http://api.ipify.org')
```
Serial output:
```
* Get host by name
*** Get socket
Allocated socket #0
* w5k socket connect, protocol=2, port=53, ip=10.0.0.1
*** Opening socket 0
* Opening W5k Socket, protocol=2
* DNS: Sending request packet...
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
Bytes avail. on sock:  265
	 * Processing 8 bytes of data
* UDP Bytes Remaining:  257
Bytes avail. on sock:  257
	 * Processing 257 bytes of data
* UDP Bytes Remaining:  0
DNS Packet Received:  b'\xaf\xcd\x81\x80\x00\x01\x00\n\x00\x00\x00\x00\x03api\x05ipify\x03org\x00\x00\x01\x00\x01\xc0\x0c\x00\x05\x00\x01\x00\x00\x03\x8f\x00\x1c\x0cnagano-19599\therokussl\x03com\x00\xc0+\x00\x05\x00\x01\x00\x00\x03\xae\x00.\x13elb097307-934924932\tus-east-1\x03elb\tamazonaws\xc0B\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x04\x17\x15\xfc\x04\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x046\xeb\xa9&\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x046\xeb\xb6\xc2\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x046\xe1\xa9\x1c\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x046\xebbx\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x04\x17\x15~B\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x046\xe1Bg\xc0S\x00\x01\x00\x01\x00\x00\x003\x00\x04\xae\x81\xd6\x14'
* DNS Answer Count:  10
* DNS ERROR: Incorrect Answer Type:  5
* DNS ERROR: Failed to resolve DNS response, retrying...
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* socket_available called with protocol 2
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
*** Closing socket #0
* Resolved IP:  -1
Traceback (most recent call last):
  File "main.py", line 66, in <module>
  File "/lib/adafruit_requests.py", line 632, in get
  File "/lib/adafruit_requests.py", line 535, in request
  File "/lib/adafruit_requests.py", line 415, in _get_socket
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 80, in getaddrinfo
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 88, in gethostbyname
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 266, in get_host_by_name
AssertionError: Failed to resolve hostname!
```
DNS server response from `dig @8.8.8.8 api.ipify.org`:
```
; <<>> DiG 9.10.3-P4-Ubuntu <<>> @8.8.8.8 api.ipify.org
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 52187
;; flags: qr rd ra; QUERY: 1, ANSWER: 10, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;api.ipify.org.			IN	A

;; ANSWER SECTION:
api.ipify.org.		2097	IN	CNAME	nagano-19599.herokussl.com.
nagano-19599.herokussl.com. 1743 IN	CNAME	elb097307-934924932.us-east-1.elb.amazonaws.com.
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 23.21.252.4
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.225.66.103
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.204.14.42
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 23.21.109.69
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.225.169.28
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.227.255.202
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.235.83.248
elb097307-934924932.us-east-1.elb.amazonaws.com. 33 IN A 54.235.182.194

;; Query time: 65 msec
;; SERVER: 8.8.8.8#53(8.8.8.8)
;; WHEN: Thu Sep 24 20:28:29 EDT 2020
;; MSG SIZE  rcvd: 268
```

brentru, 2020-09-25:
@kevinaj Are you able to try connecting to `api.ipify.org` using Arduino and the same hardware? Specifically by running this sketch: https://www.arduino.cc/en/Tutorial/LibraryExamples/WebClient.

This [library's DNS client](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/master/adafruit_wiznet5k/adafruit_wiznet5k_dns.py) shares the Arduino library's implementation. If it works there, it should work here.

kevinaj, 2020-09-26:
@brentru Just tried the WebClient Arduino sketch. Works as expected. DNS resolved to an IP and I was able to connect via HTTP.

brentru, 2020-09-28:
@kevinaj  Ok! I'll have some time this week to check this issue. DNS Answer Count looks correct, seems the error may be within parsing the answer type

kevinaj, 2020-09-28:
@brentru Thanks! As I was digging into it some more I noticed that my local DNS entries were having problems too (even just plain A records). I tracked it down to the checking of flags in the response. My local DNS server is authoritative and that bit was changing the value of `flags`. See issue #25 and pull request #26. It doesn't solve the CNAME issue, but it should at least let me test locally.

FoamyGuy, 2023-01-24:
I think this has been resolved by #73. If anyone has further issues with DNS lookups a new issue can be created to track.

closed 2023-01-24 by FoamyGuy