adafruit/Adafruit_CircuitPython_Wiznet5k #41: Buffer length check in socket_write()

bjnhur, 2021-08-03:
I think, I found a bug in **adafruit_wiznet5k/adafruit_wiznet5k.py**

line 724, the code already has `ret` variable (SOCK_SIZE check & len of buffer)
https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/f00a3f72ad20b71a71e8a590dfdf78a82f9b1e71/adafruit_wiznet5k/adafruit_wiznet5k.py#L724-L727

But, line 747, 751, didn't use this `ret` variable.
https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/f00a3f72ad20b71a71e8a590dfdf78a82f9b1e71/adafruit_wiznet5k/adafruit_wiznet5k.py#L747-L751

Need to fix? My suggestion is:
```
        # update sn_tx_wr to the value + data size
        ptr = (ptr + ret) & 0xFFFF # <-- use ret
        self._write_sntx_wr(socket_num, ptr)

        cntl_byte = 0x14 + (socket_num << 5)
        txbuf = buffer[:ret] # <- use ret
        self.write(dst_addr, cntl_byte, txbuf) # <- use txbuf
```

brentru, 2021-08-04:
@bjnhur  Would you be interested in opening a pull request? I'll take a look if you tag me.

tekktrik, 2022-02-23:
Resolved by #42 with suggested fix!