adafruit/Adafruit_CircuitPython_Wiznet5k #43: wrong offset address in _write_socket()

bjnhur, 2021-08-10:
current code

https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/f00a3f72ad20b71a71e8a590dfdf78a82f9b1e71/adafruit_wiznet5k/adafruit_wiznet5k.py#L866-L868

I think, don't need to add `base` addr like **_read_socket()** function. It's enough to use offset.

My suggestion is
```
 cntl_byte = (sock << 5) + 0x0C
 return self.write(address, cntl_byte, data)
```

[image]

brentru, 2021-08-10:
@bjnhur  I think you're correct as well, would you mind opening a PR? I can test this on the W5500 ethernet wing to verify.

bjnhur, 2021-08-11:
> @bjnhur I think you're correct as well, would you mind opening a PR? I can test this on the W5500 ethernet wing to verify.

Good. I also added this commit in current PR #42

anecdata, 2022-03-02:
Closing, fixed by #42.