adafruit/Adafruit_CircuitPython_Wiznet5k #47: example upload

min-hs, 2021-11-16:
I want an example to be added to the Adafruit_wiznet5k library.
Is it possible to upload the Adafruit library as in the examples on Wiznet's github?
For your information, they are based on Adafruit's examples. Please check if it's possible.
Link - [https://github.com/Wiznet/RP2040-HAT-CircuitPython/tree/master/examples](https://github.com/Wiznet/RP2040-HAT-CircuitPython/tree/master/examples)

brentru, 2021-11-16:
Hi, 

Feel free to submit any examples as a pull request: https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/pulls

anecdata, 2021-11-16:
A PR for any changes to support W5100S are also very welcome.

brentru, 2021-11-17:
@anecdata  @min-hs  submitted a PR (https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/pull/48) which includes support for the W5100S!

anecdata, 2021-11-17:
@brentru Very nice, saw that earlier. I'm getting WIZnet W5100S-EVB-Pico next week, so hopefully can test it out.

anecdata, 2023-01-28:
Does this need to stay open? Anyone interested can create a PR to add examples to this repo.