adafruit/Adafruit_CircuitPython_Wiznet5k #57: WIZNET5K can throw AssertionError for failing to configure DHCP

chabala, 2022-04-10:
There's an open issue, #53 about failing to connect with DHCP, but that is about constantly failing, and I don't want to hijack it.

I recently had this happen:
```
code.py output:
starting...
Traceback (most recent call last):
  File "code.py", line 29, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 207, in __init__
AssertionError: Failed to configure DHCP Server!

Code done running.
```

This is new for me, as DHCP usually works, and I was testing code by having it run for hours unattended, and having it soft reset the board when it reached its success state. DHCP was back to working again when I restarted the program.

The code in `adafruit_wiznet5k.py`:
https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/450c52e3b5a1f830581d8668cf57e779309d0aa7/adafruit_wiznet5k/adafruit_wiznet5k.py#L198-L203

These are the last lines of the `__init__` block for the `WIZNET5K` object. While failing to configure DHCP is not ideal, it's also not assertion worthy. It's calling `set_dhcp()` which has a timeout and return codes, so failure was an expected outcome.

The real issue here is throwing AssertionError for something that is not invariant; failures happen, this should raise an exception, and probably incorporate some retries first, but definitely not make any assertions.

The current user workaround to prevent hanging here would be to `try...except AssertionError` around constructing `WIZNET5K`, which is disappointing.

FoamyGuy, 2023-01-24:
The assertion has been removed and this raises a RuntimeError now. I believe that closes this issue.

Feel free to re-open if I've misunderstood and this issue still remains.

closed 2023-01-24 by FoamyGuy

chabala, 2023-01-24:
I agree, fixed by #85