adafruit/Adafruit_CircuitPython_Wiznet5k #59: Wiznet5k SSL support

Kun-Narra, 2022-05-26:
`  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 235, in connect
AssertionError: Error: SSL/TLS is not currently supported by CircuitPython.`

Is there anyone or any plan to add SSL in CircuitPython?

brentru, 2022-05-27:
There is a native SSL module in CircuitPython, https://docs.circuitpython.org/en/latest/shared-bindings/ssl/index.html, for chips that support it like the ESP32-Sx.

However, the Wiznet WIZ5500 chip used on the Adafruit Ethernet FeatherWing does not support TLS/SSL.

closed 2022-05-27 by brentru

JetForMe, 2022-06-27:
> However, the Wiznet WIZ5500 chip used on the Adafruit Ethernet FeatherWing does not support TLS/SSL.

I don’t believe this to be true. Here’s an example [library](https://github.com/WIZnet-ioLibrary/W5x00-TLS) that provides TLS on W5x00 modules, including the W5500.

TLS operates on layers 4-7, whereas TCP/IP is layer 3. TLS should be implementable on top of the existing W5500 library(ies).

Without TLS and SSL support, Ethernet connectivity is relegated to hobbyists and experimenters, and can't reasonably be used practically.

ladyada, 2022-06-28:
hiya if you are serious about helping to add tls support, open up an issue on the circuitpython repo - of course, it is not trivial.

JetForMe, 2022-06-30:
@ladyada Done!

AndreasTheCat, 2024-04-24:
@JetForMe 
Are you still interested in getting TLS/SSL to W5500 on CP? 

Kindly,
Let me know

anecdata, 2024-04-24:
Follow https://github.com/adafruit/circuitpython/pull/8954 for progress on this. TLS TCP clients (e.g., HTTPS) are working now.

JetForMe, 2024-04-24:
> @JetForMe Are you still interested in getting TLS/SSL to W5500 on CP?
> 
> Kindly, Let me know

It sounds like this work is almost complete…?