adafruit/Adafruit_CircuitPython_Wiznet5k #65: AssertionError: Failed to initialize WIZnet module.

one0410, 2022-08-30:
Hi,

I am trying to use example on Readme page. And I only change the clock,mosi,miso pin on my customized board using ESP32-S3.

```
import board
import busio
import digitalio
import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket

print("Wiznet5k WebClient Test")

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_URL = "http://api.coindesk.com/v1/bpi/currentprice/USD.json"

cs = digitalio.DigitalInOut(board.IO9)
#spi_bus = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
spi_bus = busio.SPI(board.IO36, MOSI=board.IO35, MISO=board.IO10)

# Initialize ethernet interface with DHCP
eth = WIZNET5K(spi_bus, cs)

# Initialize a requests object with a socket and ethernet interface
requests.set_socket(socket, eth)

print("Chip Version:", eth.chip)
print("MAC Address:", [hex(i) for i in eth.mac_address])
print("My IP address is:", eth.pretty_ip(eth.ip_address))
print("IP lookup adafruit.com: %s" %eth.pretty_ip(eth.get_host_by_name("adafruit.com")))

#eth._debug = True
print("Fetching text from", TEXT_URL)
r = requests.get(TEXT_URL)
print('-'*40)
print(r.text)
print('-'*40)
r.close()

print()
print("Fetching json from", JSON_URL)
r = requests.get(JSON_URL)
print('-'*40)
print(r.json())
print('-'*40)
r.close()

print("Done!")
```

I had the following response.

```
Wiznet5k WebClient Test
Traceback (most recent call last):
  File "<stdin>", line 19, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 178, in __init__
AssertionError: Failed to initialize WIZnet module.
>
```

I found that is was dead on line 393. The result always be 0x00.

https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/450c52e3b5a1f830581d8668cf57e779309d0aa7/adafruit_wiznet5k/adafruit_wiznet5k.py#L387-L394

Somebody mentioned that maybe too high frequency for SPI. No matter I change the baudrate to 100000,200000,1000000,2000000,4000000,8000000. I got the same problem.

Could somebody help me?
Thank you very much.

anecdata, 2022-08-30:
Triple-check that the SPI connections match the code, some mismatches gives this behavior. Can you be more specific about the S3 board and the WIZnet module being used?

anecdata, 2023-01-29:
@one0410 Would you mind re-testing? There have been substantial improvements to this library recently.