adafruit/Adafruit_CircuitPython_Wiznet5k #66: Can't network 2 5100s modules together

nathanmayall, 2022-09-08:
Hi there,

For an experiment I'm trying to network 2 pico devices with the Wiznet 5100s ethernet hat.

I've used the same code on both, and I'm hoping to perform an operation where they have a html page and perform get requests to each other, 

At the moment I'm configuring them to both be servers, then at a set interval they can just send a request to the other, but I can't get that far.

This issue I'm having is below, here's the debug. I'm assuming anything can be wrong, There's not much documentation out there that I can find on how to directly connect these 2 devices.

```
code.py output:
Wiznet5k WebServer Test(P2P)
My Link is: 1
Chip Version: w5100s
MAC Address: ['0x1', '0x1', '0x2', '0x3', '0x4', '0x5']
My IP address is: 192.168.2.2
w5100s
MAX_SOCK_NUM is 2
Open this IP in your browser:  192.168.2.2
*** Get socket
Allocated socket #0
* Listening on port=80, ip=192.168.2.2
*** Opening socket 0
* Opening W5k Socket, protocol=33
*** Get socket
Allocated socket #1
* Listening on port=80, ip=192.168.2.2
*** Opening socket 1
* Opening W5k Socket, protocol=33
* socket_available called on socket 0, protocol 33
Traceback (most recent call last):
  File "code.py", line 100, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_wsgiserver.py", line 95, in update_poll
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 426, in available
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 514, in socket_available
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 863, in _get_rx_rcv_size
ValueError: Invalid byteorder

Code done running.
```

And code.py

```python
import board
import busio
import digitalio
import time

import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import *
from adafruit_wsgi.wsgi_app import WSGIApp
import adafruit_wiznet5k.adafruit_wiznet5k_wsgiserver as server
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket

##SPI0
SPI0_SCK = board.GP18
SPI0_TX = board.GP19
SPI0_RX = board.GP16
SPI0_CSn = board.GP17

##reset
W5x00_RSTn = board.GP20

print("Wiznet5k WebServer Test(P2P)")

# Setup your network configuration below
# random MAC, later should change this value on your vendor ID
MY_MAC = (0x01, 0x01, 0x02, 0x03, 0x04, 0x05)
IP_ADDRESS = (192, 168, 2, 2)
SUBNET_MASK = (255, 255, 255, 0)
GATEWAY_ADDRESS = (192, 168, 2, 1)
DNS_SERVER = (192, 168, 2, 1)

led = digitalio.DigitalInOut(board.GP25)
led.direction = digitalio.Direction.OUTPUT

ethernetRst = digitalio.DigitalInOut(W5x00_RSTn)
ethernetRst.direction = digitalio.Direction.OUTPUT

# For Adafruit Ethernet FeatherWing
cs = digitalio.DigitalInOut(SPI0_CSn)

# cs = digitalio.DigitalInOut(board.D5)
spi_bus = busio.SPI(SPI0_SCK, MOSI=SPI0_TX, MISO=SPI0_RX)

# Reset W5500 first
ethernetRst.value = False
time.sleep(1)
ethernetRst.value = True

# Initialize ethernet interface without DHCP
eth = WIZNET5K(spi_bus, cs, is_dhcp=False, mac=MY_MAC, debug=True)
# Initialize ethernet interface with DHCP
# eth = WIZNET5K(spi_bus, cs, is_dhcp=True, mac=MY_MAC, debug=False)

# Set network configuration
eth.ifconfig = (IP_ADDRESS, SUBNET_MASK, GATEWAY_ADDRESS, DNS_SERVER)

print("Chip Version:", eth.chip)
print("MAC Address:", [hex(i) for i in eth.mac_address])
print("My IP address is:", eth.pretty_ip(eth.ip_address))

# Initialize a requests object with a socket and ethernet interface
requests.set_socket(socket, eth)

# Here we create our application, registering the
# following functions to be called on specific HTTP GET requests routes
web_app = WSGIApp()

html_string = """
    <!DOCTYPE html>
    <html lang="en">
    <head>
    <meta charset="UTF-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    </head>
    <body>
    sample html
    </body>
    </html>
    """

#HTTP Request handlers
@web_app.route("/")
def root(request):  # pylint: disable=unused-argument
    print("Root WSGI handler")
    # return ("200 OK", [], ["Root document"])
    return ("200 OK", [], [html_string])

# Here we setup our server, passing in our web_app as the application
server.set_interface(eth)
print(eth.chip)
wsgiServer = server.WSGIServer(80, application=web_app)

print("Open this IP in your browser: ", eth.pretty_ip(eth.ip_address))

# Start the server
wsgiServer.start()

while True:
    # Our main loop where we have the server poll for incoming requests
    wsgiServer.update_poll()
    # Maintain DHCP lease
    # eth.maintain_dhcp_lease()
    # Could do any other background tasks here, like reading sensors
```

Can anyone point me in the right direction? I don't know where to go from here.

Thanks

anecdata, 2022-09-08:
You may want to bring this to Discord and ask in #help-with-circuitpython. Are both devices running when this exception occurs? What version of CircuitPython is installed? Do these devices connect to each other through a switch or router?

nathanmayall, 2022-09-08:
@anecdata I fixed the byteorder by applying the fix in #64. Still got an error where an empty text response is received by the client though. I'll jump in discord now.

closed 2022-09-08 by nathanmayall