adafruit/Adafruit_CircuitPython_Wiznet5k #77: DNS doesn't get a response, failed to resolve hostname

Squaal, 2022-12-14:
Hi there, 

As stated, I'm having issues with a code that used to run fine at some point. DNS fails to get a response and retries until it eventually gives up. The DHCP request seems to succeed and grab an IP, but DNS fails even on the example code provided.

Hardware : Custom RP2040 board running CircuitPython 7.3.3 + Ethernet Featherwing, tried on 7.0.0 and latest W5K without success.

Below is the debug output after the DHCP lease. To my untrained eye something seems wrong about the socket having the 192.168.1.1 address on opening the W5K socket as it was assigned the 192.168.1.44 ?

The same issue happens on any example code I've found, and code that used to work.

```
* DHCP: Successful lease
*** Closing socket #0
* Found DHCP Server:
IP: bytearray(b'\xc0\xa8\x01,')
Subnet Mask: bytearray(b'\xff\xff\xff\x00')
GW Addr: bytearray(b'\xc0\xa8\x01\x01')
DNS Server: (192, 168, 1, 1)
Chip Version: w5500
MAC Address: ['0xde', '0xad', '0xbe', '0xef', '0xfe', '0xed']
My IP address is: 192.168.1.44
* Get host by name
*** Get socket
Allocated socket #0
* Listening on port=53, ip=192.168.1.44
*** Opening socket 0
* Opening W5k Socket, protocol=2
* w5k socket connect, protocol=2, port=53, ip=192.168.1.1
*** Opening socket 0
* Opening W5k Socket, protocol=2
* DNS: Sending request packet...
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* socket_available called on socket 0, protocol 2
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
```

skipping ahead

```
* DNS ERROR: Did not receive DNS response!
* DNS ERROR: Failed to resolve DNS response, retrying...
*** Closing socket #0
* Resolved IP:  -1
Traceback (most recent call last):
  File "main.py", line 110, in <module>
  File "/lib/adafruit_minimqtt/adafruit_minimqtt.py", line 474, in connect
  File "/lib/adafruit_minimqtt/adafruit_minimqtt.py", line 270, in _get_connect_socket
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 61, in getaddrinfo
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 70, in gethostbyname
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 253, in get_host_by_name
AssertionError: Failed to resolve hostname!
```

anecdata, 2022-12-15:
Can you link to the example code you mention?

Looks like DHCP is reporting 192.168.1.1 as the DNS server. I'm assuming that's your router, and that it does provide DNS.

Squaal, 2022-12-15:
Neither the code located in [wiznet5k_simpletest.py](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/main/examples/wiznet5k_simpletest.py) or in [miniMQTT](https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/blob/main/examples/ethernet/minimqtt_adafruitio_eth.py) work. Both fail to resolve hostname and both fail to get a DNS response. The router I'm using is completely open and nothing is blocking the requests. Providing the IP of the website/mqtt broker you are trying to connect to will fail as well.

I fixed the issue this morning by setting the DNS server manually to (8,8,8,8) using the _dns property. This doesn't seem like it was the case before as my code ran perfectly fine a few months ago.

anecdata, 2022-12-15:
I've been testing W5100S and W5500 WIZnet Pico boards and hats, and the Adafruit Ethernet FeatherWing on and off for months with 7.3.3 and the 8.0.0-beta releases, and the [wiznet5k_simpletest.py](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/main/examples/wiznet5k_simpletest.py) works for me. Just tested with 8.0.0-beta.5 on a QT Py ESP32-S2 with WIZnet 5100S Pico hat.

There are some open [issues](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/issues) in this library, and there may be network differences that affect results. We'd need to dig deeper to figure out why it isn't getting DNS from the router.

FoamyGuy, 2023-01-24:
I think that #73 may have resolved this. There are improvements to DNS lookups in that PR. 

If someone tries this out and is still having the same issue with current versions of circuitpython and the library feel free to re-open.

closed 2023-01-24 by FoamyGuy