adafruit/Adafruit_CircuitPython_Wiznet5k #78: dhcp fails (because firewall) but returned object from call is unusable

schaefer01, 2022-12-15:
hi, 

dhcp times out because I need the mac address for the firewall but the exception prevents the returned object from being used to pull out the mac address

 eth = WIZNET5K(spi_bus, cs)

 if the eth object contained internal state, even if there was no connection then at least I could retrieve the mac address to open the firewall for a retry.  chicken and egg.

schaefer01, 2022-12-15:
hi,
 I found the "default" mac address at the top of wiznet5k.py.
 Isn't this supposed to be unique per hardware element?
 Are there maybe a million raspberry pi ethernet modules out in the world with the same mac address?

anecdata, 2022-12-15:
Many ethernet modules don't have an embedded MAC address, so you just need to choose one that will be unique on your local network.

schaefer01, 2022-12-15:
sorry, I deleted my post when I thought I misread what I was reading in the code.

I thought uniqueness was part of the standard, I guess its not.

anecdata, 2022-12-15:
They do need a MAC address to get on the network, but often it's a separate chip providing the uniqueness. Not needed in a hobbyist environment. In a commercial environment, a more thorough circuit with managed-unique MAC addresses can be used.

schaefer01, 2022-12-15:
Adafruit products provide too much value (bang for the buck) to stay in the hobby domain, MIT Haystack is just starting to use featherwings in the lab as the core of new lab equipment, and I'm the guinea pig.

e28eta, 2023-01-24:
@schaefer01 it looks to me like you can provide any MAC you want inside the `WIZNET5K` constructor, and if I understand your use case correctly, that's all you need.

https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/73bb5b6bfeb89f00fe90502ad4804b346a7d17f3/adafruit_wiznet5k/adafruit_wiznet5k.py#L148-L154

I see a hardcoded default, `DE:AD:BE:EF:FE:ED`, in the same file: https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/73bb5b6bfeb89f00fe90502ad4804b346a7d17f3/adafruit_wiznet5k/adafruit_wiznet5k.py#L128-L129

schaefer01, 2023-01-24:
thank you, that's what I did, but not I could not use "any" address.
We have an IT department and they confiigured our LAN to block unknown mac addresses, so I had to coordinate with them first.

closed 2023-01-24 by schaefer01