adafruit/Adafruit_CircuitPython_Wiznet5k #79: dhcp.py returns 0 value in parse_dhcp_response(), causes dhcp_state_machine() to raise exception

schaefer01, 2022-12-15:
hi,
  I've got a firewall issue, parse_dhcp_response() fails and returns 0, but dhcp_state_machine() does not
   test for error response, assumes a value uses the value in a conditional statement, and raises this exception:

 object with buffer protocol required

the code is here:
        elif self._dhcp_state == STATE_DHCP_REQUEST:
            if self._sock.available():
                if self._debug:
                    print("* DHCP: Parsing ACK")
                msg_type, xid = self.parse_dhcp_response()
                # Check if transaction ID matches, otherwise it may be
                # for another device
                if htonl(self._transaction_id) == int.from_bytes(xid, "big"):    <------- exception raised here

schaefer01, 2022-12-15:
as a side note, there are mutiple different error paths in  parse_dhcp_response() that all return 0's
it would be nice to have a diagnostic to indicate which failure path is taken when a failure occurs

anecdata, 2022-12-17:
@schaefer01 Can you re-test the code from PR 80 and confirm it solves your issue?

schaefer01, 2022-12-19:
hi,
   yes, thank you.
bob s.

closed 2022-12-19 by schaefer01