adafruit/Adafruit_CircuitPython_Wiznet5k #90: adafruit_wiznet5k_socket.py has hardcoded 1000 second timeouts

e28eta, 2023-01-21:
Related, but separate, to #89 

There are at least two hardcoded timeouts for 1000:
https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/blob/73bb5b6bfeb89f00fe90502ad4804b346a7d17f3/adafruit_wiznet5k/adafruit_wiznet5k_socket.py#L178-L189

I believe `time.monotonic()` returns [the number of seconds](https://github.com/adafruit/circuitpython/issues/1846#issuecomment-489667558). Waiting 1,000 seconds (over 16 minutes) in the `__exit__` block is _far_ too long.

tekktrik, 2023-01-23:
Yeah, that does seem extremely long.  @dhalbert, I know we're exploring (or actively) changing the socket libraries.  Do you think it's worth waiting to change this with that work, or go ahead and add a fix/improvement in the meantime?

FoamyGuy, 2023-01-24:
fwiw I think we should accept a fix in the meantime if one gets submitted. 

My assumption is that this was intended to be 1000ms or 1 second but the units got mixed up. I intend to test it out with lower values this week. Will submit a PR if testing is successful and no one else has submitted yet.

BiffoBear, 2023-02-15:
The default hardware timeout settings mean that the hardware will timeout after 19 seconds. This will make a good stopgap. Eventually, polling the hardware timeout register will be a better alternative because it will respect any changes to the hardware timeout settings made by the user.

I'll open a PR with 19 seconds.

closed 2023-03-20 by FoamyGuy