adafruit/Adafruit_CircuitPython_Wiznet5k #92: socket.py accept() status() weird behavior, might be the decorator, or I might be seeing things.

schaefer01, 2023-02-01:
I'm trying again from my previous closed issue with better information and a more humble attitude

The function status() is a decorator, when used inside accept(), the parenthesis is not there.
the accept does not work for me (feather rp2040) because status does not update.
When I added parentheses to "status" (which was where I was working on when I confused myself and canceled my last "issue),python raises syntax issues, probably because a side-effect of how the decorator works. Continuing on, realizing that perhaps the decorator aspect was not doing what it should be doing.
 I created a new function named Status() identical to status() but without the decorator symbol,
 I called Status() inside of accept() instead of "status" and the accept() call started working.

Can someone verify that there is an issue with status and the decorator, and I'm not just seeing things?
thank you,
 bob s.

BiffoBear, 2023-02-11:
Hi @schaefer01,

Please will you post some example code? This will allow me to better understand your issue and see if I am able to help.

Thanks!

schaefer01, 2023-02-13:
I'll send a copy of the now working version winzip sockets.
The testing code (the other side of the socket) is python socket code running on a mac (client_socket.py).
the 2040 app is mqtt_ether.py
The IP address used on the 2040 was the value returned by a call to dhcp (home comcast or work mit), 
  the port was the default mqtt port, though it was a socket and not mqtt

the purpose of the code that I had to modify was, if the 2040's mqtt broker was the wrong IP address, the mqtt connect would time out, the 2040 would switch to listening on its own socket IP and
 I would command (via mac) the 2040 to accept a different mqtt broker (mqtt_ether.py)

On the 2040 I replaced "status()" with "Status()"
    def Status(self) -> int:
        """
        Return the status of the socket.

        :return int: Status of the socket.
        """
        return _the_interface.socket_status(self.socknum)[0]

[wiznet5k_socket.py.zip]
(https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/files/10722882/wiznet5k_socket.py.zip)

[client_socket.py.zip](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/files/10722909/client_socket.py.zip)

[mqtt_ether.py.zip](https://github.com/adafruit/Adafruit_CircuitPython_Wiznet5k/files/10722950/mqtt_ether.py.zip)

BiffoBear, 2023-02-13:
Hi Bob, thanks for getting back to me with the files. I'm mystified as to why `Status()` works and `status` does not. I'll look into it but it will take a few days.

```
    @property
    def status(self) -> int:
        """
        Return the status of the socket.

        :return int: Status of the socket.
        """
        return _the_interface.socket_status(self.socknum)[0]
```
 The `@property` decorator means that the status function should be called with `x = socket.status` and not `x = socket.status()` so the original code in `socket.accept` looks correct at first glance.

Cheers,

Martin

schaefer01, 2023-02-13:
hi,
 My app wasn't working so I changed a whole bunch of things, the last thing I changed before the app started working was this.  I also did an undo (because I couldn't believe the solution) and the bug came back. If you can't recreate the problem, I'll assume the real bug lived somewhere in cached code and my last change followed by reboot cleaned the cache, and then when I followed with an undo I then introduced a different bug, and assumed the two tests were related to my good actions and not independent related to my bad (mistaken) actions.

BiffoBear, 2023-02-17:
Hi Bob,
I've tried the `Wiznet5K` example `wiznet5k_simpleserver.py` and it works as it should. This is the code, slightly modified to work on my RPI2040 with a Wiznet 5100s:

```
import board
import busio
import digitalio
import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket

print("Wiznet5k SimpleServer Test")

cs = digitalio.DigitalInOut(board.GP17)
spi_bus = busio.SPI(board.GP18, MOSI=board.GP19, MISO=board.GP16)
eth = WIZNET5K(spi_bus, cs, is_dhcp=False, debug=True)

# (ip_address, subnet_mask, gateway_address, dns_server)
eth.ifconfig = (eth.unpretty_ip("192.168.11.3"),
                eth.unpretty_ip("255.255.255.0"),
                eth.unpretty_ip("192.168.11.1"),
                (8, 8, 8, 8),)
             
# Initialize a socket for our server
socket.set_interface(eth)
server = socket.socket()  # Allocate socket for the server
server_ip = eth.pretty_ip(eth.ip_address)  # IP address of server
server_port = 50007  # Port to listen on
server.bind((server_ip, server_port))  # Bind to IP and Port
server.listen()  # Begin listening for incoming clients

while True:
    print(f"Accepting connections on {server_ip}:{server_port}")
    conn, addr = server.accept()  # Wait for a connection from a client.
    print(f"Connection accepted from {addr}, reading exactly 1024 bytes from client")
    with conn:
        data = conn.recv(1024)
        if data:  # Wait for receiving data
            print(data)
            conn.send(data)  # Echo message back to client
    print("Connection closed")
```

Perhaps you could try to run a simple example like this to confirm whether `socket.accept()` is working on your system?

Sorry that I took so long to get back to you.

Cheers,

Martin

schaefer01, 2023-02-17:
thank you, must have been me.