adafruit/Adafruit_CircuitPython_Wiznet5k #102: Fixed socket timeout

tkcstms, 2023-03-01:
I had an issue (and a few hours of figuring out what was going on) with the HTTP server example.
It always loaded for several seconds before anything appeared in the web browser.
I figured out when creating an instance of a WSGIServer, the socket timeout was hard coded to be 20 seconds.
I don't know if I have some error on my side but I always had to wait 20 seconds before the request has been answered.

I modified the constructor of WSGIServer to fit my needs and now it accepts a float parameter to set the socket timeout and it works perfectly for me.

Please feel free to correct me if I missed something!

`Modified constructor of WSGIServer

    def __init__(
        self,
        port: int = 80,
        debug: bool = False,
        timeout: float = 1,
        application: Optional[callable] = None,
    ) -> None:
        """
        :param int port: WSGI server port, defaults to 80.
        :param bool debug: Enable debugging, defaults to False.
        :param Optional[callable] application: Application to call in response to a HTTP request.
        """
        self.application = application
        self.port = port
        self._timeout = timeout
        self._client_sock = []
        self._debug = debug

        self._response_status = None
        self._response_headers = []
        if _the_interface.chip == "w5100s":
            self.MAX_SOCK_NUM = const(2)
        else:
            self.MAX_SOCK_NUM = const(6)
        if self._debug:
            print("Max sockets: ", self.MAX_SOCK_NUM)`

fieldOfView, 2023-04-20:
The real question is why the timeout happens _always_ on consecutive requests.

kocinski15, 2023-06-02:
I noticed exactly the same. When running WSGIServer on RPI2 with Wiznet 5100 Hat , every request took 20 secs to answer.