adafruit/Adafruit_CircuitPython_Wiznet5k #103: socket is blocking after some time or with multiple requests on W5100S pico

IakovlevAA, 2023-03-07:
I've created this [Issue](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/41), when trying to create HTTP server on Wiznet5k Pico W5100S. Unfortunately, I could solve this only particularly, because socket isn't working stable. It was blocking, if not receivinig bytes after some not fixed time, or receiving multiple requests at once. I've managed to solve double request(it was HTTP lib problem) and time blocking, but 3 or more requests blocking socket immediately.

BiffoBear, 2023-04-07:
Hi, Please could you post your code and the error trace from the serial out? I'll have a look at it and see if there is a fix.

IakovlevAA, 2023-04-07:
hi, i am using this https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/41#issue-1611035373, traceback is the same. Code stucks in "write" method, because there are multiple calls to "listen" or "accept", then it stucks in 
**adafruit_wiznet5k.py**
```
# if buffer is available, start the transfer
        free_size = self._get_tx_free_size(socket_num)
        while free_size < ret:
            free_size = self._get_tx_free_size(socket_num)
            status = self.socket_status(socket_num)[0]
            if status not in (SNSR_SOCK_ESTABLISHED, SNSR_SOCK_CLOSE_WAIT) or (
                timeout and time.monotonic() - stamp > timeout
            ):
                ret = 0
                break
```
because it can't get free size of socket_num

IakovlevAA, 2023-04-17:
The topic has no connection to the actual version of socket lib. Closing as it was HTTP lib problem