adafruit/Adafruit_CircuitPython_Wiznet5k #104: [Feature Request] Wiznet "w6100-evb-pico"

KHome, 2023-04-02:
Hi,

please check to support additional to the W5100 and W5500 as well the W6100.
I've bought this board: https://docs.wiznet.io/Product/iEthernet/W6100/w6100-evb-pico

Because the W5100s is pin-2-pin compatible to W6100, I've used this Circuitpython version:
Adafruit CircuitPython 8.0.5 on 2023-03-31; W5100S-EVB-Pico with rp2040

>>> %Run -c $EDITOR_CONTENT
My editor has this caode:
```
import board
import busio
import digitalio
import time
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K

##SPI0
SPI0_SCK = board.GP18
SPI0_TX = board.GP19
SPI0_RX = board.GP16
SPI0_CSn = board.GP17

##reset
W5x00_RSTn = board.GP20

print("Wiznet5k Ping Test (no DHCP)")

# Setup your network configuration below
# random MAC, later should change this value on your vendor ID
MY_MAC = (0x00, 0x01, 0x02, 0x03, 0x04, 0x5)
IP_ADDRESS = (192, 168, 178, 100)
SUBNET_MASK = (255, 255, 255, 0)
GATEWAY_ADDRESS = (192, 168, 178, 1)
DNS_SERVER = (8, 8, 8, 8)

led = digitalio.DigitalInOut(board.GP25)
led.direction = digitalio.Direction.OUTPUT

ethernetRst = digitalio.DigitalInOut(W5x00_RSTn)
ethernetRst.direction = digitalio.Direction.OUTPUT

# For Adafruit Ethernet FeatherWing
cs = digitalio.DigitalInOut(SPI0_CSn)
# For Particle Ethernet FeatherWing
# cs = digitalio.DigitalInOut(board.D5)
spi_bus = busio.SPI(SPI0_SCK, MOSI=SPI0_TX, MISO=SPI0_RX)

# Reset W5500 first
ethernetRst.value = False
time.sleep(1)
ethernetRst.value = True

# Initialize ethernet interface with DHCP
# eth = WIZNET5K(spi_bus, cs)
# Initialize ethernet interface without DHCP
eth = WIZNET5K(spi_bus, cs, is_dhcp=False, mac=MY_MAC)
```
which throws this error message
```
Wiznet5k Ping Test (no DHCP)
Traceback (most recent call last):
  File "<stdin>", line 47, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 195, in __init__
RuntimeError: Failed to initialize WIZnet module.
```

Please support.

BiffoBear, 2023-04-06:
Hi,

While the 6100 is pin to pin compatible with the 5100, their internal firmware are very different. The 6100 supports IPv4 and IPv6 with 6 ports. The 5100 is IPv4 only with 4 ports. This means that the CircuitPython Wiznet5K driver will have to be heavily modified to support the 6100 chip, even in IPv4 mode.

I've got the same 6100 module on order, so that I can look at possibly adding support for the Wiznet 6100. However, I won't have the module unit mid May and I have limited time to devote to working on the Wiznet5K driver.

If (big if) I were to add support for the 6100 to the Wiznet5K driver it would initially only be for IPv4 so there would be no benefit over the 5xxxx series in the medium term.

That being said, someone else may step up and write a Wiznet6K driver.

TLDR: If you need a CP compatible IPv4 Ethernet module in the near term then your best bets are the 5100 (4 ports) and 5500 (8 ports) as they are already supported.

HTH.

KHome, 2023-04-11:
Thank you for your feedback. As I bough three W6100-EVB-PICO, I will wait until these little thingies will be supported. In the meantime I will proceed with the 5xxxx series.

BiffoBear, 2023-06-16:
@KHome Hello, I hope this message finds you well. I've got a pull request in to add w6100 IPv4 support to the WIZNET5K library. If you know your way around GitHub and can test the PR, I would be grateful.

Thx!

KHome, 2023-06-16:
Hi, Thanks for your Upgrade.
I will check today at night.

KHome, 2023-06-24:
Yeah! It works :-)

I've used this firmware:
https://circuitpython.org/board/wiznet_w5500_evb_pico/

replaced the provided DHCP examples, like `wiznet5k_simpletest.py` with this PIO config:
`
SPI0_SCK = board.GP18
SPI0_TX = board.GP19
SPI0_RX = board.GP16
SPI0_CSn = board.GP17

cs = digitalio.DigitalInOut(SPI0_CSn)
spi_bus = busio.SPI(SPI0_SCK, MOSI=SPI0_TX, MISO=SPI0_RX)`

The board is finally working as expected. 
@BiffoBear: Thank you very much !