adafruit/Adafruit_CircuitPython_Wiznet5k #126: support for socketpool?

schaefer01, 2023-06-28:
Hi,
  I'd like to use adafruit_ntp but this library has a dependency on socketpool, which isn't supported for the rp2040 featherwing (which I am using) because the featherwing doesn't have a wifi, hence I'm using the wiznet5k ethernet module. 
So, I'm asking you to please look into the effort it would take to not only allow the import of wiznet5k_socket but also the import of wiznet5k_socketpool.
Thank  you,
  bob s.

anecdata, 2023-06-28:
I wasn't sure if it would work, but it seems to:
```py
import time
import board
import busio
import digitalio
import adafruit_ntp
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket
from secrets import secrets

cs = digitalio.DigitalInOut(board.GP17)
spi = busio.SPI(board.GP18, MOSI=board.GP19, MISO=board.GP16)
eth = WIZNET5K(spi, cs, debug=False)
eth_ip = eth.pretty_ip(eth.ip_address)
print(f"WIZnet Client IP Address: {eth_ip}")

ntp = adafruit_ntp.NTP(socket, tz_offset=-5)

while True:
    print(ntp.datetime)
    time.sleep(5)
```
Result:
```
code.py output:
WIZnet Client IP Address: 192.168.6.250
struct_time(tm_year=2023, tm_mon=6, tm_mday=28, tm_hour=12, tm_min=2, tm_sec=36, tm_wday=2, tm_yday=179, tm_isdst=-1)
struct_time(tm_year=2023, tm_mon=6, tm_mday=28, tm_hour=12, tm_min=2, tm_sec=41, tm_wday=2, tm_yday=179, tm_isdst=-1)
```

schaefer01, 2023-06-28:
hi,
  I just tried this configured for my board, this is great!!
  thank you so much,
bob s.

closed 2023-06-28 by schaefer01

anecdata, 2023-06-29:
`socket.set_interface(eth)` is needed for use with v3.0.0 of this library