adafruit/Adafruit_CircuitPython_Wiznet5k #142: RuntimeError: Failed to initialize WIZnet module

JTL48, 2024-02-01:
I am having issues connecting an Ethernet Featherwing with a Feather RP2040 microcontroller. I used the "adafruit-circuitpython-adafruit_feather_rp2040-en_US-8.2.9.uf2" (Latest Stable) file to image the RP2040. I added adafruit_wiznet5k, adafruit_bus_device, and adafruit_requests.mpy to the lib folder. I got these from the "adafruit-circuitpython-bundle-8.x-mpy-20240118" bundle. 

When running wiznet5k_simpletest_manual_network.py, it gives this error:
`eth = WIZNET5K(spi_bus, cs)`

```
Traceback (most recent call last):
  File "code.py", line 48, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 243, in __init__
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 1059, in _wiznet_chip_init
RuntimeError: Failed to initialize WIZnet module.
```
I tried messing with the Baurdrates. Nothing worked there. I tried putting every library file in that bundle in the lib folder just in case. I tried getting the most updated version of Adafruit_CircuitPython_Wiznet5k and Adafruit_CircuitPython_BusDevice. Now I have the wiznet5k .py files instead of .mpy files on the board so I can debug further. I found that this is where it's getting caught:
```
def _detect_and_reset_w5500() -> bool:
    ...
    self._write_mr(0x08)
    if self._read_mr() != 0x01:
         return False
    ...
```

Do you have any suggestions on a fix for this? Any help would be appreciated.

anecdata, 2024-02-01:
Can you post the code, and is this the default wiring on the Ethernet FeatherWing (`D10` I believe lines up with the default `CS`)?

There is a debug mode that may give you more info: `adafruit_wiznet5k.adafruit_wiznet5k.WIZNET5K( ..., debug=True)`

JTL48, 2024-02-02:
Here is the code I am using:

```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import busio
import digitalio
import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"

# Setup your network configuration below
IP_ADDRESS = (192, 168, 0, 71)
SUBNET_MASK = (255, 255, 255, 0)
GATEWAY_ADDRESS = (192, 168, 0, 1)
DNS_SERVER = (0, 0, 0, 0)

print("Wiznet5k WebClient Test (no DHCP)")

rst_eth = digitalio.DigitalInOut(board.D25)
rst_eth.direction = digitalio.Direction.OUTPUT

cs = digitalio.DigitalInOut(board.D10)
cs.direction = digitalio.Direction.OUTPUT
spi_bus = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)

#print("CS Value: ", cs.value)

#print("SPI Baudrate: ", spi_bus.frequency)

# Initialize ethernet interface without DHCP

#eth = WIZNET5K(spi_bus, cs, is_dhcp=False)
eth = WIZNET5K(spi_bus=spi_bus, cs=cs, reset=rst_eth, is_dhcp=False, debug=True)

# Set network configuration
eth.ifconfig = (IP_ADDRESS, SUBNET_MASK, GATEWAY_ADDRESS, DNS_SERVER)

# Initialize a requests object with a socket and ethernet interface
requests.set_socket(socket, eth)

print("Chip Version:", eth.chip)
print("MAC Address:", [hex(i) for i in eth.mac_address])
print("My IP address is:", eth.pretty_ip(eth.ip_address))
print(
    "IP lookup adafruit.com: %s" % eth.pretty_ip(eth.get_host_by_name("adafruit.com"))
)

# eth._debug = True
print("Fetching text from", TEXT_URL)
r = requests.get(TEXT_URL)
print("-" * 40)
print(r.text)
print("-" * 40)
r.close()

print()
```

I am using debug=True and it doesn't seem to change any messaging.
And yes, D10 lines up with the CS pin.

anecdata, 2024-02-02:
I'm not sure what else to suggest. Your code works on my 8.2.9 Feather RP2040 + Ethernet Featherwing. I have some differences, not sure why they'd make a difference:
• different CS pin
• `rst_eth = None` (no RST pin wired up)
• changed static IP info to match my network

JTL48, 2024-02-02:
Weird thing is we bought 2 featherwings and both have the same issue. I thought there was a pretty low chance both would have a hardware problem.

Not sure if pictures will help at all but here are some:

[image]

[image]

[image]

anecdata, 2024-02-02:
I'm baffled too, there aren't a lot of variables here. Do you have any other SPI-based FeatherWings (or something else that can be manually wired up SPI) to try to rule out the RP2040?

Are you using the latest 5.0.6 library version (not sure what date that came out "two weeks ago")? Though it doesn't look like those changes would affect this issue.

anecdata, 2024-02-02:
I'm not seeing this line: `if self._read_mr() != 0x01:` in the library code, either the 20240118 or current version. Line 1059 is something else.

anecdata, 2024-02-02:
They look legit. Micro Center is reputable. Amazon I suppose it depends on the seller.

Can you reconcile the differences in the checks on register values that we're seeing?

JTL48, 2024-02-02:
Oh yeah for the library code, I tried to add debug statements to the decompiled .py file:

```
def _detect_and_reset_w5500() -> bool:
            """
            Detect and reset a W5500 chip. Called at startup to initialize the
            interface hardware.

            :return bool: True if a W5500 chip is detected, False if not.
            """
            print("Trying to detect W5500")
            self._chip_type = "w5500"

            print("Trying to reset 5x00")
            if not self._sw_reset_5x00():
                return False

            print("Trying to write mr 0x01")
            self._write_mr(0x01)
            if self._read_mr() != 0x01:
                return False

            print("Trying to write mr 0x10")
            self._write_mr(0x10)
            if self._read_mr() != 0x10:
                return False

            print("Trying to write mr 0x00")
            self._write_mr(0x00)
            if self._read_mr() != 0x00:
                return False

            print("Trying to read versionr")
            if self._read(_REG_VERSIONR[self._chip_type], 0x00)[0] != 0x04:
                return False
            # Initialize w5500
            _setup_sockets()
            return True
```

I just changed that 0x08 value to 0x01 to test if it would respond differently with a different value. And I guess when I copied it over to this forum I forgot to change one of the values back to 8.

But yeah the outputs would look like this:
```
Trying to detect W5500
Trying to reset 5x00
Trying to write mr 0x01
Traceback (most recent call last):
  File "code.py", line 48, in <module>
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 243, in __init__
  File "/lib/adafruit_wiznet5k/adafruit_wiznet5k.py", line 1059, in _wiznet_chip_init
RuntimeError: Failed to initialize WIZnet module.
```

JTL48, 2024-02-02:
But to clarify, I’ve tested with the .mpy files from the bundle, the most up to date .mpy files from GitHub, then I tested the .py files with those comments first with the original 0x08 register value and then tried changing it to 0x01. All of this with the same result.

caternuson, 2024-02-02:
Are the headers on the Feather also soldered?

JTL48, 2024-02-02:
Yes. So I soldered the stacking headers to the RP2040: 

https://www.adafruit.com/product/2830?gad_source=1&gclid=Cj0KCQiAwvKtBhDrARIsAJj-kTg9rGjjVJaKDuS2p9hin9LcvRValu_FLnFyMEbvtcPyKl1xk91wKpEaAoX5EALw_wcB

And then I soldered normal male header pins to the Ethernet Featherwing.

JTL48, 2024-02-02:
Again this probably doesn’t help much but here’s some pictures:

[image]

[image]

[image]

[image]

anecdata, 2024-02-03:
The `CS` jumper on the bottom of the Ethernet FeatherWing is still intact? 
[image]