adafruit/Adafruit_CircuitPython_Wiznet5k #170: Unexpected connections from 0.0.0.0:0 on ESP32 Ethernet Server

niclas-gr, 2024-10-03:
## Hardware Setup

- Adafruit ESP32 Feather V2
- Adafruit Ethernet FeatherWing

## Issue Description

When setting up a server on the ESP32 using the Ethernet FeatherWing, the socket immediately and repeatedly accepts connections from a non-existent client with IP address `0.0.0.0` on port `0`. This occurs before any actual client attempts to connect, requiring continuous closing of these phantom connections.

## Minimal Working Example (MWE)

```python
import os
import time
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import busio
import digitalio
import board
from adafruit_connection_manager import get_radio_socketpool

IP_ADDRESS = (192, 168, 1, 100)
SUBNET_MASK = (255, 255, 0, 0)
GATEWAY_ADDRESS = (192, 168, 1, 1)
DNS_SERVER = (0, 0, 0, 0)
PORT = 5000

spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.D33)
eth = WIZNET5K(spi, cs, is_dhcp=False, debug=False)
eth.ifconfig = (IP_ADDRESS, SUBNET_MASK, GATEWAY_ADDRESS, DNS_SERVER)

pool = get_radio_socketpool(eth)
sock = pool.socket(pool.AF_INET, pool.SOCK_STREAM)
sock.bind((str(eth.pretty_ip(eth.ip_address)), PORT))
sock.listen()

while True:
    client_sock, addr = sock.accept()
    print(f"{time.monotonic()}: accepted connection from {addr}")
    
    if addr == ("0.0.0.0", 0):
        client_sock.close()
        continue
    
    client_sock.close()
```

## Observed Behavior

The first five print statements consistently show:
```
26.816: accepted connection from ('0.0.0.0', 0)
26.856: accepted connection from ('0.0.0.0', 0)
26.899: accepted connection from ('0.0.0.0', 0)
26.941: accepted connection from ('0.0.0.0', 0)
26.979: accepted connection from ('0.0.0.0', 0)
```

## Expected Behavior

The server should only accept connections from actual clients attempting to connect, not from the non-existent `0.0.0.0:0`.

## Questions

1. Is this a known issue with the WIZnet5k library or the Ethernet FeatherWing?
2. Could this be related to how the Ethernet interface is initialized or configured?
3. Are there any workarounds or solutions to prevent these phantom connections?

Any insights or suggestions on how to resolve this issue would be greatly appreciated. Thank you!

anecdata, 2024-10-03:
What version of CircuitPython?

I doubt it's hardware-specific to the FeatherWing, most WIZnet hardware behaves similarly, especially all of the W5500--based hardware.

niclas-gr, 2024-10-03:
I use CircuitPython 9.1.1, adafruit_wiznet5k 7.0.0, and adafruit_connection_manager 3.1.0.

dhugoexe, 2025-02-26:
Same issue, is anybody find a solution ?

lifeRobot, 2025-05-09:
RP2040 has the same issue, and when calling recv to read data, it directly blocks the asyncio library
CircuitPython version 9.2.4
adafruit_wiznet5k version 7.2.2

see picture:
[image]