adafruit/Adafruit_CircuitPython_Wiznet5k #173: RuntimeError : pystack exhausted

stanelie, 2024-12-13:
I've started trying to use the W5500 interface ot my Waveshare ESP32-S3-ETH to do some artnet stuff (lighting over network). Upon initializing a big array, I encounter the error "pystack exhausted".

I require this array and I need to manipulate it to talk to other devices that do lighting on the network. For debugging purposes, I'm not even touching the array, just creating it, and it errors out. If I remove the array, my test code runs fine. I use this same array with a similar python script that runs wifi instead of Wiznet5k, and it also runs fine.

I'm running `Adafruit CircuitPython 9.2.1 on 2024-11-20; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3` (not the correct firmware for my device, but it's the one I've found that matches it the best, since my device does not yet have a specific CircuitPython firmware.

Thoughts?

Here is my code that errors out : 
```
import time
import traceback
import board
import digitalio
import busio
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socketpool as socketpool

# For ESP32-S3-ETH
cs = digitalio.DigitalInOut(board.GPIO14)
rst = digitalio.DigitalInOut(board.GPIO9)
spi = busio.SPI(board.GPIO13, MOSI=board.GPIO11, MISO=board.GPIO12)
eth = WIZNET5K(spi, cs, reset=rst, is_dhcp=True, debug=False)

pool = socketpool.SocketPool(eth)

HOST = ""
PORT = 6454
TIMEOUT = None
MAXBUF = 10
reply_array = [
                0x41, 0x72, 0x74, 0x2d, 0x4e, 0x65, 0x74, 0x00, # "Art-Net\0" header
                0x00, 0x21,                                     # opcode
                0x02, 0x00, 0x00, 0x01,                         # our IP address
                0x36, 0x19,                                     # port 6454
                0x00, 0x01,                                     # version info of this node
                0x00,                                           # NetSwitch
                0x00,                                           # SubSwitch (universe?)
                0x2B, 0xFA,                                     # OEM code (Ex Machina is 2BFA)
                0x00,                                           # UBEA version #
                0xF0,                                           # status
                0x00, 0x00,                                     # ESTA manufacturer code
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, # Short Name 18 bytes
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00,
                0x00, 0x00, 0x00, 0x00,               
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x80, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x50, 0xc2, 0x37, 0x14, 0x24,             # our mac address
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
                0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
                ]

with pool.socket(pool.AF_INET, pool.SOCK_DGRAM) as s:
    s.settimeout(TIMEOUT)
    s.bind((HOST, PORT))
    print(f"Listening on {eth.pretty_ip(eth.ip_address)}, {PORT}")

    buf = bytearray(MAXBUF)
    while True:
        try:
            size, addr = s.recvfrom_into(buf, MAXBUF)
            msg = bytes(buf)
            print(f"received {msg}")
            s.sendto(msg, (addr[0],PORT) )
            print(f"sent {msg}")

        except Exception as ex:
            traceback.print_exception(ex, ex, ex.__traceback__)
            time.sleep(1)
```

Here are the error messages : 
```
Traceback (most recent call last):
  File "<stdin>", line 13, in <module>
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 265, in __init__
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 279, in set_dhcp
  File "adafruit_wiznet5k/adafruit_wiznet5k_dhcp.py", line 186, in request_dhcp_lease
  File "adafruit_wiznet5k/adafruit_wiznet5k_dhcp.py", line 441, in _dhcp_state_machine
  File "adafruit_wiznet5k/adafruit_wiznet5k_dhcp.py", line 202, in _dsm_reset
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 501, in ifconfig
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 1123, in _write
RuntimeError: pystack exhausted
```

stanelie, 2024-12-13:
Nevermind, I found the answer.
I changed my array to this format instead :
```
reply_array = (
                b'\x41' b'\x72' b'\x74' b'\x2d' b'\x4e' b'\x65' b'\x74' b'\x00' # "Art-Net\0" header
                b'\x00' b'\x21'                                     # opcode
                b'\x02' b'\x00' b'\x00' b'\x01'                         # our IP address
                b'\x36' b'\x19'                                     # port 6454
                b'\x00' b'\x01'                                     # version info of this node
                b'\x00'                                           # NetSwitch
                b'\x00'                                           # SubSwitch (universe?)
                b'\x2B' b'\xFA'                                     # OEM code (Ex Machina is 2BFA)
                b'\x00'                                           # UBEA version #
                b'\xF0'                                           # status
                b'\x00' b'\x00'                                     # ESTA manufacturer code
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' # Short Name 18 bytes
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00'               
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x01' b'\x80' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x80' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x50' b'\xc2' b'\x37' b'\x14' b'\x24'             # our mac address
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00' b'\x00'
                )
```

According to this post https://www.reddit.com/r/circuitpython/comments/rntq67/rookie_question/ , 
`instead of a = [0x01, 0x02] you should use: a =( b'\x01' b'\x02') that way you use 8bits array instead a int array that uses 32bits.`

I did not know that.

closed 2024-12-13 by stanelie

anecdata, 2024-12-13:
Also by using a tuple (`()`) instead of a list (`[]`), you save memory.

stanelie, 2024-12-13:
I did not know that either, newbie I am!
Thanks again!