adafruit/Adafruit_CircuitPython_Wiznet5k #174: link LED status reversed

stanelie, 2024-12-13:
This is more a cosmetic issue than anything, but the link status LED of my Waveshare ESP32-S3-ETH is reversed. It lights up when I disconnect the ethernet cable.

This seems unrelated to the Wiznet5k library, since it does this even if I do not use the library, but since it is related to this chip, I thought I would post this here. Is there something that initializes it when CircuitPython boots? Or did Waveshare made a mistake with the wiring of the LED?

anecdata, 2024-12-14:
I don't have one of these boards yet, and it's not entirely clear to me from the schematic: Is there a separate link status LED on the board, or is this the green LED typically built into the magnetics of the RJ-45 connector?

In the W5500 chip datasheet ([p.9/67](http://docs.wiznet.io/img/products/w5500/W5500_ds_v110e.pdf)), LINKLED levels are spec'd as:
> Link LED
This shows the Link status.
Low: Link is established
High: Link is not established

In the Waveshare ESP32-S3-ETH [schematic](https://files.waveshare.com/wiki/ESP32-S3-ETH/ESP32-S3-ETH-Schematic.pdf), the LINKLED signal passes through a resistor and the LED, then to ground. So a high signal is needed to light up the LEDs.

However, in the WIZnet [W55RP20-EVB-PICO](http://docs.wiznet.io/Product/ioNIC/W55RP20/w55rp20-evb-pico) implementation , the LINKLED signal connects to the cathode of a separate LED and a resistor, then to 3v3. So a low signal is needed to light up the LED. There is also the green LED in the connector, and it seems to be wired similarly (checking a similar part datasheet). I have this board and it behaves as expected: LINKLED is lit (separate LED and RJ-45 connector) when the cable connects the board to an active switch.

If I read all of that right, Waveshare for whatever reason implemented it opposite of how WIZnet did (at least in the case of that one recent board).

But in any case, this is all in hardware and not under control of any CircuitPython firmware or library.

anecdata, 2024-12-14:
I see now, there is a separate LED:
[image]
Presumably the green LED on there RJ-45 connector lights up when connected, even if the separate LED turns off?

stanelie, 2024-12-14:
On my board, the green activity led is always on, and it blinks wneh there is network activity.

The yellow link led is on at boot, but it turns off when I connect the ethernet port to another ethernet device.if I disconnect the ethernet cable, it turns back ON.

anecdata, 2024-12-14:
The RJ-45 connector link LED, and the separate board link LED, are always opposite in state?

It _could_ have been intentional, but it is inconsistent with expectations. Anyway, closing this, but feel free to re-open if new info points to the library.

closed 2024-12-14 by anecdata

stanelie, 2024-12-14:
There is no connector LEDs. There are only the board LEDs.